svenhb/GRBL-Plotter #442: Automatic pen change.

Hostaler, 2025-04-17:
Good afternoon.

I've been learning GRBL-Plotter for many days.

I've watched many of the author's videos on YouTube.
I've focused more on automatic tool change (to add to my Neje3 laser). I'm looking into the possibility of adding a linear tool changer, so I only have to program the opening and closing of the magnet.

I've seen that my Neje3 has its own GRBL version V2.2.3. And for my "laser" (milling machine, tangential cutting, digital pen writing, and hopefully soon pencil writing and automatic tool changer), I find it much easier to open the magnet with M9 and close the magnet with M8.

I've read all the scripts (I don't know much about computers, or almost nothing), but I don't know where to put M8 to activate the magnet and M9 to deactivate it.

Thank you in advance for your help.

Thank you very much and have a nice day, everyone.

P.S. The video of the electromagnet is from another author and I don't see it appropriate to publish it here.

svenhb, 2025-04-17:
Which example is the closest to your needs? https://grbl-plotter.de/index.php?id=setups&setlang=en
Probably https://grbl-plotter.de/index.php?id=auto-pen-change so on the bottom of the page you find the script for taking and removing a pen...

Hostaler, 2025-04-17:
Without a doubt, this one: https://grbl-plotter.de/index.php?id=auto-pen-change  I think I've seen it many times but hadn't realized it.
Thank you so much for your kindness and congratulations on your program.

Hostaler, 2025-04-17:
> ¿Qué ejemplo se ajusta mejor a tus necesidades? https://grbl-plotter.de/index.php?id=setups&setlang=en Probablemente https://grbl-plotter.de/index.php?id=auto-pen-change . En la parte inferior de la página encontrarás el script para tomar y retirar un bolígrafo.

I have GRBL-Plotter installed for all users in C:\Program Files (x86)\GRBL-Plotter and it creates the folder: user-public files-GRBL-Plotter-scripts or something like that. If I or someone else changes the script, for example on the moon, does it affect me?

svenhb, 2025-04-17:
Any script will be loaded, when needed. Yes, if someone changes your script, it will affect your tool-change (for example).
But you could store your script in your local folder and load it from there, but still someone could change the setting in the setup-window. But you could export your machine settings to your local storage and import each time, you need the machine,

Hostaler, 2025-04-17:
Thank you so much for your kindness.