svenhb/GRBL-Plotter #443: Line Tracing

Stoli123, 2025-06-07:
In an older version, when I chose to Create G-Code from Image, I could select a mode that started at the top and moved down line by line. I assume that is "Line Tracing". When I select that mode, it seems to do the same thing that Vectorization does. Is that by design?

svenhb, 2025-06-07:
The working prinziples are different (Form GCode from Image https://grbl-plotter.de/index.php?id=form-image&setlang=en):
In 'Line tracing' one by the other image pixel will be processed - line by line, in spiral pattern or other pattern to generate Z or S value. Better use 'B/W grayscale mode'.
In 'Vectorization' pixel with same color will be collected in shaped which can be filled. REcommend with 'Color mode'.

Via setup, you can define 'Auromatic vectorization' (https://grbl-plotter.de/index.php?id=form-setup-1#format-related):
Loaded image will be vectorized directly, without opening any form for setup. Using different algorithm for vectorization.
Only doing black/white. Was made for this kind of art: https://youtu.be/3JnVT8EVs3w

svenhb, 2025-06-07:
In B/W mode it still works top-down:

[image]

[image]

[image]

Stoli123, 2025-06-07:
OK. Thanks. I noticed that but it varied the Z height based on the gray scale value and I'm needing an On/Off value for Z. Basically either ON (Z-3.0) or OFF (Z2.00). I'll explore those settings when I get a chance. Thanks for your time. I'm always amazed at your quick response!!!

svenhb, 2025-06-07:
For on/off, just convert the image to black and white before creating the code (b/w or dithering):

[image]

[image]

Stoli123, 2025-06-10:
Thanks again. Another question regarding Sorting. I'm importing an SVG file with multiple objects (paths) that are filled, so I am applying a hatch fill.  Is there a combination of settings that would plot each object (outline and fill) in it's entirety from upper left to lower right? I can set the initial start point, but some objects are skipped, or only partially drawn and are finished after everything else is drawn.

svenhb, 2025-06-11:
The sorting algorithm is somewhat simple: From the initial starting point, it searches for the nearest starting point of a figure, at the end of this figure it searches for the nearest starting point of the next figure. 

You could use grouping, to keep objects together (if they use same color, pen-width or layer) https://grbl-plotter.de/index.php?id=form-setup-1#grouping