svenhb/GRBL-Plotter #447: Set G54 after color change.

Hostaler, 2025-07-27:
Hello, good afternoon. I've tried every way I dont know to configure the program to activate G54 after each tool change. I haven't found a solution, or I don't know how to find it.

 Therefore, I've decided to add G54 to the end of each script ( Greifen (script_pen2_select.nc) and Ablegen (script_pen2_remove.nc) ) after G90.
 I don't think an extraction function would be necessary; just confirming that it's feasible is enough.

                                               Thank you very much..

[image_flying_devil_.txt](https://github.com/user-attachments/files/21453452/image_flying_devil_.txt)

[image]

svenhb, 2025-07-27:
Yes, should work, if the tool table is activated here:

[image]

Hostaler, 2025-07-27:
WOW! 
 I don't know how to express this, your GRBL plotter just has to talk to itself. 

 You had it all planned out. I think you've worked hard on CNC, GRBL, CAM design, or... it's a mystery, but you've already thought it all out.

 I also have some good news: GRBL-Ploter is working fine with non-Arduino GRBL. (https://github.com/svenhb/GRBL-Plotter/issues/441) I'll keep you updated. 

                                                 Thanks a lot for everything.

Hostaler, 2025-07-27:
Don't worry too much (although it's easy to say, but not to do, since the same thing happens to me) since I'm teaching kids with limited resources about almost all types of industrial machining machines. I also want to launch a CNC-based product for the laser engravers that already exist, but I don't have much time.

 I'll close this issue next week, since I don't have the machine at my side right now.

                 Thanks, Svenhb.

closed 2025-07-31 by Hostaler